xperiments/p1touch #1: No response from screen...

rinytin, 2023-07-28:
Love the idea. And The software installed, Connected to my P1S, But the screen after calibration, Just has no response. When touched nothing. I tested touch before installing the software, Worked perfect. Not sure why its froze.

xperiments, 2023-07-28:
Have you setup the WiFi etc? Btw not sure if something has changed between
P1P firmware. I am out for vacation so I will not able to test it until
some weeks

xperiments, 2023-07-28:
Also, are y using the 2.8 resistive screen or other model?
Capacitive/Resistive need different drivers...

xperiments, 2023-08-03:
> Yes, Using the 2.8 screen. But Here is this latest, On every boot or power
> cycle, It wants me to rerun the P1Touch network setup. And it connects,
> Allows you to use the screen for 2 clicks or 2 touches to the screen and it
> freezes. Wonder if i got a bad screen. Enjoy Your Vacation.

Umm really strange then, can you try to use another power source? It seems
to me that this can be the problem source as others w the same board don't
have reported this problem... If you have it connected directly to the
computer then makes sense... Can you try and give some feedback... Thanks

Btw as new firmware is out I will be unable to test things until back from
vacation

rinytin, 2023-08-09:
Just a little update, It was the cable for power, Must have been too thin, But Just when we power cycle and or turn off the P1S, It looses its config and i have to reconfigure the AP and printer information. Once you open the code, I know a lot of use will help and source to larger screens.

xperiments, 2023-08-10:
How are you powering the screen? It needs to be powered from the internal
usb power to don't lost config on every power cycle. If not will continue
checking for the P1P and after some retries will clean the config  data.
also remember is only for local lan

xperiments, 2023-08-23:
Duplicate of #4

closed 2023-08-23 by xperiments