xperiments/p1touch #2: Where is source code for p1touch.bin

cjsieh, 2023-07-28:
I would like to use this on the 3.5 version of this board.  Need source code to make that happen.

Thanks

xperiments, 2023-07-28:
I will open the code when I get home from vacation... Will be first
September week.. no P1P with me...

andypiper, 2023-08-08:
FWIW this is at least partially working on the X1C (currently running some tests to see what might be missing), would love to see the source and help to contribute here, when you are back and able to share it.

TackleBerryCrash, 2023-08-10:
Config reset on sleep is a show stopper for this but this is a great start.

psiberfunk, 2023-08-27:
> Why would you want to use this on the x1 ?

jonnywatchbox, 2023-09-03:
did anyone get the 3.5 screen working?

Milkman337, 2023-09-12:
Any updates will the source be uploaded now its september

nurtext, 2023-09-13:
Friendly reminder from my side, too. I'd like to have a look at the source and compile it myself. The project would heavily benefit from releasing the source though...

nurtext, 2023-09-18:
@xperiments Any news on this topic?

xperiments, 2023-09-19:
Yes, I have a real Live... When it's published it's published. Don't ask more...

nurtext, 2023-09-20:
No need to be salty.

TackleBerryCrash, 2023-09-20:
no need to be antsy, needy, rude.... 

It will be here when it will be here. Beta testing continues. You can move along if you cant wait.

psiberfunk, 2023-09-20:
I would generally say yes.. but "antsy" is fair with closed source things you put on your network .  I, and many others can't use this until I can see the source and verify it's not laced with phone-home malware.

TackleBerryCrash, 2023-09-20:
Yes, you are right it was a resonabnle request and answer was given numberous times no need to beat the dead horse again and again and being borderline rude but at the same time nobody is forcing this to anyone. 
anyone not trusting this ( I would not obviously if wasnt seen the code itself and collaborated) can stroll past this repo. 
again no one is forcing anyone to install this. I know first hand dev is really pouring into this for weeks.

he is being rude and you know he is being rude too :) 

I think code will be released soon.

psiberfunk, 2023-09-20:
> Yes, you are right it was a resonabnle request and answer was given numberous times no need to beat the dead horse again and again and being borderline rude but at the same time nobody is forcing this to anyone. anyone not trusting this ( I would not obviously if wasnt seen the code itself and collaborated) can stroll past this repo. again no one is forcing anyone to install this. I know first hand dev is really pouring into this for weeks.
> 
> he is being rude and you know he is being rude too :)
> 
> I think code will be released soon but my answer is it will be released when it is released.

Perhaps I interpret it differently.  I didn't take it as rude... but pestering perhaps :).  Sometimes people just need reminders is all.  Anyhow, i'm sure we can all move on peaceably without a flame war.

TackleBerryCrash, 2023-09-20:
I am too old for flame wars TBH :)

nurtext, 2023-09-20:
@TackleBerryCrash If I was meant: I didn't want to be rude and I don't consider myself a rude guy. I was just asking, because @xperiments told anyone asking the source to be released in the first week of september. 

However, @xperiments behavior isn't very professionell nor polite. I mean: Yes, he's a private person, but he decided to go public with his work the moment he released the binary blob as well as instructions. There's also a lot going on on Printables regarding housings for the touchscreen. So I guess the downloads are much higher than the commentors or software devs checking this repo regularely. 

My final thoughs on this are: If you decide to publish something and making promises for releasing the source code: Do it or don't. I even don't know why he closes the issue, because it hasn't been resolved so far and the request remains.

TackleBerryCrash, 2023-09-20:
Sir, you can move on, it will be released when released.
be more professional than him and code your own and release it yourself.
you had a valid concern but pestering the dev is not cool.

Lets end this here. 
Wait if you can if not toodaloo, bon voyage,  auf wiedersehen....

Issue is closed b/c answer is given but you dont get the answer apparently

xperiments, 2023-09-21:
Code will be published next week, we are currently checking beta feedback.

@nurtext yes I have been a little rude, I am sorry
@psiberfunk @nurtext I see your points and are normal but this does not make me unprofessional. As commented I have a life, the plan was to release on vacation return but life... 

BTW I see things another way.. if you needed in a hurry to see the code just DM me as @TackleBerryCrash done. Him has been able to access code since some time...

Milkman337, 2023-09-23:
quick question where can I DM you @xperiments

xperiments, 2023-09-23:
@xps3riments / @xperiments in twitter

DIYLAB-DE, 2023-09-29:
I don't use Twitter - may I still have a look at your sources in an appropriate way? 
Best, Bruno

Milkman337, 2023-09-30:
I also dont have a twitter and would like to take a look too

xperiments, 2023-09-30:
This monday will be open

Arnos-Kramkiste, 2023-10-01:
any newer bin file or beta versions to test ? my screen turn off after some seconds and so on

xperiments, 2023-10-02:
https://github.com/xperiments-in/xtouch

closed 2023-10-02 by xperiments