xperiments/p1touch #4: Asks to configure after every shut down.

ZachWSays, 2023-08-06:
I have this plugged into the printer and after every shut down it asks me to connect to the Wi-Fi and configure before it's usable.

php4umagento, 2023-08-07:
I second that - I think it is trying to check if internet connection is alive and if it is not - it is wiping out configuration - this is show stopper

rinytin, 2023-08-08:
I 3rd that....

andypiper, 2023-08-09:
I think there is some kind of issue with the way this is persisting the configuration.

In the serial log I'm seeing this at start:

```
[P1PTouch][FS] LITTLEFS Mount SUCCESSFUL.
[P1PTouch][FS] Checking for touch config
[   501][E][vfs_api.cpp:105] open(): /littlefs/config.json does not exist, no permits for creation
*wm:AutoConnect 
*wm:No wifi saved, skipping 
```

Looks like the `config.json` cannot be opened/created for some reason. Hopefully once the source code is shared we can look into this.

TackleBerryCrash, 2023-08-10:
I fourth this. I was wondering what was happening, this clears it up.

xperiments, 2023-08-10:
Yeah you need to power it from the printer, so power cycle it affects also
the screen, if not is doing same process you comment. It tries to
reconnect, after some iterations it resets config. Also it needs local lan
mode

TackleBerryCrash, 2023-08-10:
Can you iterate "it needs local lan mode" ? Device have to be in LAN ONLY MODE for this to work ?

andypiper, 2023-08-10:
It does not (should not), as far as I know.

xperiments, 2023-08-10:
Yeah, it has not been tested in cloud mode so yes only works in lan mode
only

Anyway makes no sense to make it work for cloud as is closed source and
then the bambu mobile app makes more sense..

Anyway I will check in the future but with low priority.. mostly because is
closed source.

TackleBerryCrash, 2023-08-10:
I am confused, what is closed source ? FW is yes, but whole thing works over MQTT and that does not care if LAN mode on or off. 
screen connects to the LAN IP of the printer nor WAN so it should work fine on both modes

TackleBerryCrash, 2023-08-10:
Like I said,  I am using home assistant on all my BL printers and that works over MQTT locally on cloud enabled printers which it does not really care since it is connecting thru the LAN IP

TackleBerryCrash, 2023-08-10:
Actually, I can even connect the screen to my p1p from outside wifi or a mobile hotspot thru NAT/Port forwarding. Network layer will care less about where it is connected thru. It will only care if there is an answering MQTT client on the other end.
This should work not dependent on what mode printer is on, as it is really irrelevant at that layer.

andypiper, 2023-08-11:
Works (in some ways) with my X1C with cloud mode enabled, but with the P1Touch connecting over the LAN. Should not be an issue.

TackleBerryCrash, 2023-08-16:
well, update I have tried all the possible cases except one but I doubt it is it. (newest p1p firmware has a mqtt fix.)
Screen sleep does not cause the config revert, but after an hour or so it gives the above command and literally panics and reverts.
LAN only or Cloud, all same.

So my take is maybe panic is due to when it deep sleeps it loses wifi connection and after wake up it panics before it was able to reconnect. Could be a race condition

iBlack22, 2023-08-16:
Ich habe das gleiche Problem Sobald das Display kein Strom mehr hat und es dann wieder eingeschalten wird sind alle Einstellungen nicht mehr vorhanden

TackleBerryCrash, 2023-08-16:
Powering from printer or externally results the same fate. It races before printer can answer the MQTT queries then when there is no answer, it panics and reverts

redlinebikeparts, 2023-08-20:
LAN Only or cloud has the same result where the screen will ask to be reconfigured when powered off for a time this is regardless of if its being powered by printer or externally. 

We also started a group for this amazing mod. 

https://m.facebook.com/groups/284441287626603/?ref=share&mibextid=S66gvF

dezign999, 2023-08-20:
Looking at console logs, it appears that the entire config is deleted if the P1Touch cannot connect to the P1 after 122+/- seconds. Once it erases the settings, it attempts to reconnect and by not having any saved settings it bounces you back to the initial setup process. This resetSettings should probably be commented out at the very least, but here's where it happens:

`[P1PTouch][MQTT] ---- CONNECTION FAIL ----: -2
*wm:resetSettings 
*wm:SETTINGS ERASED 
[P1PTouch][FS] Delete config
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13192
load:0x40080400,len:3028
entry 0x400805e4
[P1PTouch][FS] LITTLEFS Mount SUCCESSFUL.
[P1PTouch][FS] Checking for touch config
[   554][E][vfs_api.cpp:105] open(): /littlefs/config.json does not exist, no permits for creation
*wm:AutoConnect 
*wm:No wifi saved, skipping 
*wm:AutoConnect: FAILED for  122 ms
*wm:StartAP with SSID:  P1PTouch
*wm:AP IP address: 192.168.4.1
*wm:Starting Web Portal `

yezzer, 2023-08-20:
So is there currently no way around this? It's by design? If we turn off the printer at the wall overnight, everything is reset?

Can I suggest instead that reflashing the firmware if we want to reset everything might be a better option?

xperiments, 2023-08-20:
It's not from design, it's a bug XD.. I will fix it when I get home.

I will try a better path to configure it where resetting will be done
manually from the screen

yezzer, 2023-08-20:
> It's not from design, it's a bug XD.. I will fix it when I get home.
> 

Ah ok great 😁 Thanks for your work on this, looks great and super useful!

dezign999, 2023-08-20:
> It's not from design, it's a bug XD.. I will fix it when I get home. I will try a better path to configure it where resetting will be done manually from the screen
> […](#)

Cheers! A “Try again” and “Reset” buttons might be one way to go. Enjoy your vacay!

azsenca, 2023-08-27:
Is source code available for screen firmware, so more people can contribute and get things going faster?

drakengard780, 2023-09-10:
i confirm lost settings if connect to usb of printer and shutdown the printer. every time reconfigure settings require