xperiments/p1touch #5: Allow changing screen orientation

andypiper, 2023-08-09:
I found [a great case for the ESP32-2432S028R](https://www.printables.com/model/310352-alien3d-deck) which would be perfect for this project! Unfortunately, it expects the screen to be oriented with the USB cable at the top left, instead of the bottom right.

Could a configuration option be added to rotate the screen by 180 degrees?

ZachWSays, 2023-08-09:
Watch the video on the printables page for that, seems that it can be slid in either way. Should be able to just do it upside down.

andypiper, 2023-08-09:
Yes, I do have it set that way currently, but it means that a couple of minor things are aligned differently (SD slot on the bottom, which is not currently used in p1touch (unless that's the secret of the missing config file in #4 - I think not as that refers to vfs / LittleFS rather than SD, but I could be mistaken). Otherwise, yes, it remains an excellent option either way up, and I'm being needy suggesting the screen rotation :-)

xperiments, 2023-08-10:
You can refer to the printable one Lee published.. but not mean to be final
as I am moving the design to be on top and be the  base for the next
versions

fdebeer, 2023-09-09:
use a longer usb-cable, or use the wires that come with the board. solder a connector to these wires to feed the board.

xperiments, 2023-09-09:
LCD Flip is added for next version

closed 2023-09-09 by xperiments

smford, 2023-09-09:
> use a longer usb-cable, or use the wires that come with the board. solder a connector to these wires to feed the board. on facebook there are examples how to do this

What about other people who don't have this level of skill or tools?

Wouldn't it simply be simply easier, to use the most trivial method of doing it via software, so that it solves the problem for everyone once and for all? 

Which is a button that toggles the lvgl orientation within the settings menu, and saves it to config.

Thank you @xperiments