xperiments/p1touch #8: how can i change colour ?

CalvinLeong8208, 2023-08-21:
my screen white and purple colour ... didnt look normal .... try to reinstall again .. still a same ... why like that ?

CalvinLeong8208, 2023-08-21:
[image]
[image]
mine esp32 come with type-c and mirco usb ... mirco usb didnt work with pc .. but type-c can connect to pc ... mean mirco cant flash the rom ... side panels and ATM ? didnt get you ... newbie here ....

xperiments, 2023-08-21:
Umm seems a new model. I will buy one when back. Mine has only microusb but
I bought it last year.. btw is yours 2.8 or 3.5 inch?

itsonlymedan, 2023-08-21:
This is strange I have only bought mine recently and mine is not like that at all.

xperiments, 2023-08-21:
Can you share a link or image?

CalvinLeong8208, 2023-08-22:
mine is 2.8 inch .....

closed 2023-08-22 by CalvinLeong8208

itsonlymedan, 2023-08-24:
Out of a small sample piece 32 people we have identified that only 2 people out of the 32 have USB C everyone else has micro