xperiments/p1touch #19: Wont connect

RadsWa, 2023-08-26:
Log included. 
I can install this, set the parameters in the config, but that's as far as it goes, when I press save, it goes to the connecting to p1p, and then reverts back to "connect to the p1 network to continue, I have tried it at the printer, and also right next to my modem, I can see it, it's got 5 bars, it just won't connect to my wifi (yes I triple checked all info), it actually says it connects then erases the info check log attached.

There is also some other errors in the log file.
It would be really handy if there was away to manually enter the info so the system doesn't remove it, thru something like putty.

*wm:5 networks found
[154845][E][WebServer.cpp:649] _handleRequest(): request handler not found
*wm:5 networks found
[168003][E][WebServer.cpp:649] _handleRequest(): request handler not found
*wm:7 networks found
[188118][E][WiFiClient.cpp:517] flush(): fail on fd 50, errno: 11, "No more processes"
[188121][E][WebServer.cpp:649] _handleRequest(): request handler not found
*wm:Connecting to NEW AP: Aussie Broadband 0361
*wm:connectTimeout not set, ESP waitForConnectResult... 
*wm:Connect to new AP [SUCCESS] 
*wm:Got IP Address: 
*wm:192.168.1.137 
*wm:config portal exiting 
[P1PTouch][WifiManager] WiFi CONNECTED
[P1PTouch][FS] Config Saved
[P1PTouch][WifiManager] Local IP: 192.168.1.137
[P1PTouch][MQTT] Connecting
[241077][E][ssl_client.cpp:129] start_ssl_client(): socket error on fd 48, errno: 113, "Software caused connection abort"
[241078][E][WiFiClientSecure.cpp:144] connect(): start_ssl_client: -1
[P1PTouch][MQTT] ---- CONNECTION FAIL ----: -2
*wm:resetSettings 
*wm:SETTINGS ERASED 
[P1PTouch][FS] Delete config
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13192
load:0x40080400,len:3028
entry 0x400805e4
[P1PTouch][FS] LITTLEFS Mount SUCCESSFUL.
[P1PTouch][FS] Checking for touch config
[   486][E][vfs_api.cpp:105] open(): /littlefs/config.json does not exist, no permits for creation
*wm:AutoConnect 
*wm:No wifi saved, skipping 
*wm:AutoConnect: FAILED for  122 ms
*wm:StartAP with SSID:  P1PTouch
*wm:AP IP address: 192.168.4.1
*wm:Starting Web Portal

xperiments, 2023-08-27:
Have you entered the SN all in CAPS? Thx

RadsWa, 2023-08-27:
Hi Yes I did, I even copied it directly from the serial on a closed case I
had so there would be no mistakes.
I tried many times.

fdebeer, 2023-08-28:
it seems some boards wont work. there are more with this issue. I think it is a bug in the board you have.

slowpaddle, 2023-09-14:
I bought the exact board that was linked to in the description and the board will not connect to my WiFi network either. I'm thinking there's more to it than the board itself.

fdebeer, 2023-09-18:
the board only connects to a 2.4 GHz network, so if you have a router with 5GHz network, it won't work. 
So be shure to select a 2.4 GHz network. 
All modern routers should be able to transmit  2.4 Ghz, but not all do this automaticaly. check your router for the possibilities

slowpaddle, 2023-09-18:
I checked my router and it connects automatically to whatever frequency the remote device needs. Also tried it on a straight 2.4Ghz  router and still nothing. Frustrating eh?