xperiments/p1touch #20: Unable to progress after install - No calibration + prompts

MandMBricks, 2023-08-27:
Hello P1 Friends!
Just got my screen and have tried using the online web installer to flash the screen.  Everything seems to flash fine, but I never get the "+" prompts to calibrate the screen, I just get a "P1Touch with Wifi bars" screen, which in turn does not allow me to progress through the wifi setup and further.  Looking at the log it just keeps displaying the same logs, over and over.  

"ELF file SHA256: 3c70301b3a6b5412
Rebooting...
ets Jul 29 2019 12:21:46
rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13192
load:0x40080400,len:3028
entry 0x400805e4
[P1PTouch][FS] LITTLEFS Mount SUCCESSFUL.
[P1PTouch][FS] Checking for touch config
[   421][E][vfs_api.cpp:105] open(): /littlefs/touch.json does not exist, no permits for creation
[P1PTouch][TOUCH] Touch Setup
Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.
Core  1 register dump:
PC      : 0x400e91da  PS      : 0x00060230  A0      : 0x800d28e4  A1      : 0x3ffb2140  
A2      : 0x00000000  A3      : 0x8d800000  A4      : 0x00000000  A5      : 0x00ef013f  
A6      : 0xffffffff  A7      : 0x3ffc9b58  A8      : 0x800e91da  A9      : 0x3ffb2120  
A10     : 0x000001b1  A11     : 0x00000001  A12     : 0x00000000  A13     : 0x000073e8  
A14     : 0x000001b1  A15     : 0x00000f6e  SAR     : 0x0000000a  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x00000004  LBEG    : 0x400845d5  LEND    : 0x400845dd  LCOUNT  : 0x00000027  
Backtrace: 0x400e91d7:0x3ffb2140 0x400d28e1:0x3ffb2160 0x400e79fe:0x3ffb2190 0x400db0df:0x3ffb21b0 0x400e915c:0x3ffb2200 0x400d7af4:0x3ffb2230 0x400d7f53:0x3ffb2260 0x4011140e:0x3ffb2290"

Tried looking for this issue but can't seem to find anyone that is having it.  Tried multiple time to install, both PC and MAC and no luck.  Thanks for any advice or tips

fdebeer, 2023-08-27:
RTFM or go to your phone, and search for the P1Ptouch wifinetwork. use pass 12345678 to log in. there you can configure the p1ptouch for your printer.

m33ps, 2023-08-27:
Are you sure you are using the ESP32-2432S028R board that is linked?  Did you maybe get the ESP32-3248S035?  They look almost identical.

MandMBricks, 2023-08-27:
Yeah, I tried resetting it a few times with no luck.  I did verify that I purchased the correct board.  Honestly, I'm stumped right now.

fdebeer, 2023-08-27:
No problems here with the board. but i read on FB there are more with this problem. hope for you it's not a broken board. 
as it is not a really expencive board, buy another one and see if this solves the problem.
and hopefully xperiments comes with an update for the software for the problem with losing the config after powercycle, witch also covers this problem

MandMBricks, 2023-08-27:
Luckily it's a cheap board, so if boinked, no big deal.  Hopefully there will be a solution at some point.  Thanks for the advice!

DBarr3, 2023-08-27:
So t feel bad. I have the same exact issue. Been driving me crazy. I reported it as an issue on the main page. Will probably need to scrap the idea and go with an old iPhone with the Bambu handy app

MandMBricks, 2023-08-27:
@DBarr3 I hate that!  If I come across a solution, I will be sure to come back and update.

xperiments, 2023-09-09:
Duplicate of #4

closed 2023-09-09 by xperiments