xperiments/p1touch #23: Stuck on P1TOUCH logo and a wifi sign with no response whatsoever.

bial0004, 2023-08-31:
I bought the sxreen from the link you put from aliexpress and did everything as mentioned but after the instalation is done the screen is stuck on the logo and under it only the wifi sympol with no response.
the logs page is showing this:

ELF file SHA256: 3c70301b3a6b5412

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13192
load:0x40080400,len:3028
entry 0x400805e4
[P1PTouch][FS] LITTLEFS Mount SUCCESSFUL.
[P1PTouch][FS] Checking for touch config
[   421][E][vfs_api.cpp:105] open(): /littlefs/touch.json does not exist, no permits for creation
[P1PTouch][TOUCH] Touch Setup
Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400e91da  PS      : 0x00060230  A0      : 0x800d28e4  A1      : 0x3ffb2140  
A2      : 0x00000000  A3      : 0x8d800000  A4      : 0x00000000  A5      : 0x00ef013f  
A6      : 0xffffffff  A7      : 0x3ffc9b58  A8      : 0x800e91da  A9      : 0x3ffb2120  
A10     : 0x000001b1  A11     : 0x00000001  A12     : 0x00000000  A13     : 0x00007168  
A14     : 0x000001b1  A15     : 0x00000ef8  SAR     : 0x0000000a  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x00000004  LBEG    : 0x400845d5  LEND    : 0x400845dd  LCOUNT  : 0x00000027  

Backtrace: 0x400e91d7:0x3ffb2140 0x400d28e1:0x3ffb2160 0x400e79fe:0x3ffb2190 0x400db0df:0x3ffb21b0 0x400e915c:0x3ffb2200 0x400d7af4:0x3ffb2230 0x400d7f53:0x3ffb2260 0x4011140e:0x3ffb2290

Please help!
Thanks in advanced

fdebeer, 2023-08-31:
see #11 and #20.
same problem already mentioned.
also the program is stil in beta, so expect bugs

bial0004, 2023-09-01:
> see #11 and #20. same problem already mentioned. also the program is stil in beta, so expect bugs
> 
> but did you go on your phone to the p1touch wifinetwork? if this shows up, log in with pass 12345678 and set up the wifi and printer.

Unfortunately not, The wifi network doesn't appear at all in my phone. I even tried two different phones but no success

fdebeer, 2023-09-02:
I have 4 boards now, and they all work great. 
first, see if you got the right board: mine are model: ESP32-2432S028R
working from a windows machine, installed the CH340 driver. when connecting to the board, chose com port USB serial . for me it's com 3. dont forget to erase the divice
now wait for about 2 min to install.
click the 2 crosses to calibrate the screen. dont wait to lang for this or the board resets itself.

my log is:
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13192
load:0x40080400,len:3028
entry 0x400805e4
[P1PTouch][FS] LITTLEFS Mount SUCCESSFUL.
[P1PTouch][FS] Checking for touch config
[   438][E][vfs_api.cpp:105] open(): /littlefs/config.json does not exist, no permits for creation
*wm:AutoConnect 
*wm:No wifi saved, skipping 
*wm:AutoConnect: FAILED for  119 ms
*wm:StartAP with SSID:  P1PTouch
*wm:AP IP address: 192.168.4.1
*wm:Starting Web Portal

xperiments, 2023-09-09:
Close for innactivity

closed 2023-09-09 by xperiments