xperiments/p1touch #24: Default wifi password - 12345678 to be something else

php4umagento, 2023-08-31:
Hello
 If I have neighbour who has P1Touch I can connect to his and he can connect to mine as password is same for all of us, can this be more personalised - based on MAC address or something else?

Don't want anyone to connect to my P1Touch

php4umagento, 2023-08-31:
Thanks for your great work so far, when is next release planned?

xperiments, 2023-08-31:
No ETA at the moment. I will be working on this from next week to fix the
known issues and begin cleaning the code and testing it privately until
release then source the code... But mostly 2-3 weeks

fdebeer, 2023-08-31:
> Hello If I have neighbour who has P1Touch I can connect to his and he can connect to mine as password is same for all of us, can this be more personalised - based on MAC address or something else?
> 
> Don't want anyone to connect to my P1Touch

when setup and configured and connected to your printer the wifi p1touch network is no longer available.

xperiments, 2023-09-09:
No more need of Wifi Access Point Setup in next version... Closing for now

closed 2023-09-09 by xperiments