xperiments/p1touch #35: Install problem

SzilagyiR, 2023-09-19:
How can I reset the modul old configuration?

The web installer gives error: Failed to initialize. Try resetting your device or holding the BOOT button while clicking INSTALL.
Holding the button but nothing happens, the old software runing on the screen.

Thanks!

TackleBerryCrash, 2023-09-19:
Installer erases the device prior to install. Do you see that on the screen ?

xperiments, 2023-09-19:
Seems a driver problem. Have you installed the correct drivers? if so, Can you try rebooting your computer completely and try again?

parrothd, 2023-09-19:
Same here, it was setup via the old web setup program, now it doesn't see the device when connected.

xperiments, 2023-09-19:
Can you try using this software for the first flash? 

Download the 0.8.2 firmware http://retroconsol.es/products/p1touch/ota/p1touch.0.8.2.bin

Flash it with this utility:

https://github.com/esphome/esphome-flasher/releases

SzilagyiR, 2023-09-19:
Neither the web installer nor the esp flasher not seen the com port.

The driver must be ok, the first intallation wenth smooth two weeks berfore.

TackleBerryCrash, 2023-09-19:
Mac or PC

SzilagyiR, 2023-09-19:
Pc

parrothd, 2023-09-19:
I had 2 bad cables, windows would only power on the p1 touch, not see it.

TackleBerryCrash, 2023-09-19:
open up the device manager and see what ports you have before plugging in also after pluggin in 
also dont use front IO, try back IO

TackleBerryCrash, 2023-09-19:
> I had 2 bad cables, windows would only power on the p1 touch, not see it.

oh yes, it has to be a data cable. :)

TackleBerryCrash, 2023-09-19:
what solved it ? :)

SzilagyiR, 2023-09-19:
Switching the usb cable :)

TackleBerryCrash, 2023-09-19:
I have two rules in this life 
never trust a fart and never trust a cable :)

SzilagyiR, 2023-09-19:
Thanks fo everyone, who tried to help!

closed 2023-09-19 by SzilagyiR