xperiments/p1touch #45: Setting Options Missing.

chrisping12, 2023-09-21:
I cannot find on any screen inside the P1touch where I can select OTA updates. I also cannot find any firmware.bin files anywhere on the site. 

The only item I see in setting is Aux fan and chamber fan.

Can you help?

TackleBerryCrash, 2023-09-21:
please use discord for help. This place is for issues

closed 2023-09-21 by TackleBerryCrash

chrisping12, 2023-09-21:
Can you send me the link to P1Touch discord?

chrisping12, 2023-09-21:
I never received an beta invite email. Any other way to access the discord now?

xperiments, 2023-09-21:
If you have not signup for the beta no

chrisping12, 2023-09-22:
Is there any current location I can download the most current version of the firmware? 
Or as Requested is there any possibility of joining the beta testing though I am a couple days past?