xperiments/p1touch #46: "Load Filament" function does not end gracefully, BambuLab Studio can't initiate new print

CerealPrint, 2023-09-22:
After using the "Load Filament" function BambuLab Studio cannot send a new print job. Apparently the commands sent by the P1Touch do not end the loading job gracefully.

Then the whole printer needs to power cycle to accept new jobs.

XTouch v0.8.6
P1P Firmware 01.04.00.00

en-x, 2023-09-23:
I am having this same issue. P1Touch 0.8.6 and P1S 01.04.00.00. I tried this several times.
The original P1S screen loads and unloads find.

xperiments, 2023-09-28:
Fixed in 0.8.17. Can you check. I close meanwhile. Reopen if needed

closed 2023-09-28 by xperiments

en-x, 2023-09-28:
Seems to working perfectly now. Thank you.
By the way, great job with his project. I am loving it.

CerealPrint, 2023-09-28:
Tested it again with XTouch v0.8.17 and Bambu P1P 01.04.01.00 and it seems to work now.