xperiments/p1touch #52: Filament on holder may be tangled error everytime

jmc71479, 2023-09-26:
I keep getting 12FF800000020001: The filament on the spool holder may be tangled by or stuck,” with every new print, but it runs. It happens when printer says cloud slice queue, I think, right before it downloads to printer.  Error stays on screen until acknowledged.  Happened with v0.8.6, and I updated to v0.8.14 tonight and is still happening.  This is on P1P with P1S upgrades, Firmware v. 01.04.00.08, Printing from Bambu Studio v. 1.7.6.92. I tried to attach video of it happening, but this doesn’t seem to allow video.  Image of error attached.
[image]

TackleBerryCrash, 2023-09-26:
Can you check the HMS on the printer ? Do you see the error there also  ?

jmc71479, 2023-09-26:
I don’t see any errors on the printer itself when this happens, or on Bambu studio.

Jeremy M. Carroll

xperiments, 2023-09-28:
Yeah, I was reporting a level that bambustudio does not report. 
Will fix

xperiments, 2023-09-28:
Fixed in 0.8.17. Can you check. I close meanwhile. Reopen if needed

closed 2023-09-28 by xperiments