xperiments/p1touch #54: DOES NOT LOAD THE FIRMWARE

tpverticaljm, 2023-09-26:
Using 'COM6' as serial port.
Connecting....
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting....
Detecting chip type... ESP32
Connecting....

Chip Info:
 - Chip Family: ESP32
 - Chip Model: ESP32-D0WD-V3 (revision 3)
 - Number of Cores: 2
 - Max CPU Frequency: 240MHz
 - Has Bluetooth: YES
 - Has Embedded Flash: NO
 - Has Factory-Calibrated ADC: YES
 - MAC Address: 08:D1:F9:ED:97:B8
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
 - Flash Size: 4MB
Unexpected error: The firmware binary is invalid (magic byte=FF, should be E9)

TackleBerryCrash, 2023-09-26:
Please update your serial driver and try with another cable, also baud rate needs to be 115200

Please use discord for support. You maybe using wrong screen too.

this is not a bug so I am closing the issue

closed 2023-09-26 by TackleBerryCrash

xperiments, 2023-09-26:
But how are y flashing the device?