xperiments/p1touch #55: el firmware no se instala

tpverticaljm, 2023-09-26:
parece que no gusta que os digan que las cosas van mal y si no es así... lo de ayudar tampoco es lo vuestro.

el firmware no se instala, he intentado de cientos de manera incluso con espflash y os mandé los errores pero en vez de guiarme decidieron cerrar el post, gracias.

cuando acudo aqui es porque ya he probado como 10 cables, 3 pantallas y tropocientas formas de instalar el firmware y ninguna ha funcionado.

si no quieren responder por aqui y dirigen a discord o cualquier otro medio, almenos peguen el enlace.

gracias.

xperiments, 2023-09-26:
Hola,

Pues por el tono, esperarás a release. Dado que el grupo de discord es privado por ahora. Y como ya lo de ayudar no es lo nuestro pues te esperas.

closed 2023-09-26 by xperiments

tpverticaljm, 2023-09-27:
a los hechos me remito...