xperiments/p1touch #59: Pair Printers have no way to connect this

3PrintD-Solution, 2023-09-30:
Unfortunately I can't pair my 2 P1Ps. The green button has no function.

LOG

`[XTouch][SD] Reading JSON file: /xtouch/printer.json
[XTouch][SD] Reading JSON file: /xtouch/printer-ips.json
[XTouch][SD] Writting JSON file: /xtouch/printer-ips.json
[XTouch][SD] Reading JSON file: /xtouch/printer.json
[XTouch][SD] Reading JSON file: /xtouch/printer-ips.json
[XTouch][SD] Writting JSON file: /xtouch/printer-ips.json
[XTouch][SD] Reading JSON file: /xtouch/printer.json
[XTouch][SD] Reading JSON file: /xtouch/printer-ips.json
[XTouch][SD] Writting JSON file: /xtouch/printer-ips.json
[XTouch][SD] Reading JSON file: /xtouch/printer.json
[XTouch][SD] Reading JSON file: /xtouch/printer-ips.json
[XTouch][SD] Writting JSON file: /xtouch/printer-ips.json
[XTouch][SD] Reading JSON file: /xtouch/printer.json
[XTouch][SD] Reading JSON file: /xtouch/printer-ips.json
[XTouch][SD] Writting JSON file: /xtouch/printer-ips.json
[XTouch][SD] Reading JSON file: /xtouch/printer.json
[XTouch][SD] Reading JSON file: /xtouch/printer-ips.json
[XTouch][SD] Writting JSON file: /xtouch/printer-ips.json
[XTouch][SD] Reading JSON file: /xtouch/printer.jso`

That's all that happens

[image]
[image]

xperiments, 2023-09-30:
Please contact us in Discord. We can try to help yo there

3PrintD-Solution, 2023-09-30:
Thanks. The Discord Name?

xperiments, 2023-09-30:
If you have access to the beta. You have the invitation in an email. I
can't give you directions here. Or DM me directly in twitter @xps3riments

xperiments, 2023-10-01:
Closed as was a touch miss configuration

closed 2023-10-01 by xperiments