xperiments/p1touch #61: P1 touch and homeassitant coexistence

Pitax, 2023-10-01:
I tested the public  (that was available on the web configurator) version and I noted that if the P1touch is connected to the printer I lose the connection to home assistant.
Is this a limit of the MQTT broker of the printer, do you think there is a solution to this problem?

Maybe use an external MQTT broker for both the printer or the P1touch? I saw in some home assistant forum that you can "bridge" the MQTT broker on the printer to have the same data on another broker.

xperiments, 2023-10-01:
https://github.com/xperiments/p1touch#p1ps-01040000-lan-mode-bug

ATM is not in roadmap to solve it. Will wait until fw 01.05. Adding more complexity w a bridge is not my idea. Connecting via bbl login... maybe in future...

closed 2023-10-01 by xperiments

Arnos-Kramkiste, 2023-10-01:
> I tested the public (that was available on the web configurator) version and I noted that if the P1touch is connected to the printer I lose the connection to home assistant. Is this a limit of the [...]

use new firmware on p1p or p1s version 01.04.01.00 is out.